MIT-PSFC/disruption-py #331: OSSFE 2025

RemDelaporteMathurin, 2024-10-04:
Congrats on open-sourcing this!

Would you consider submitting an abstract at the Open Source Software for Fusion Energy conference?

See https://ossfe.github.io/

@allen-adastra @gtrevisan

gtrevisan, 2024-10-04:
thanks, we'll take a look!

closed 2024-10-04 by gtrevisan

allen-adastra, 2024-10-04:
I'm not a main developer of disruption-py; def all Greg and crew 🙂

I'd be happy to submit an abstract on POPSIM; assuming you are okay with me presenting software that is "supposed to be open source eventually".

Cheers,
Allen

RemDelaporteMathurin, 2024-10-04:
> assuming you are okay with me presenting software that is "supposed to be open source eventually".

I think the committee would prefer avoiding this as often we heard that and things never got released ;-) Better if there's already an Open Source licence