MIT-PSFC/disruption-py #346: Add option to not run a specific method

AlexSaperstein, 2024-11-14:
when specifying the RetrievalSettings, it would be useful to be able to specify specific methods to not be run. the advantage of this would be that you can still specify "all" for the run tags, and then specifically pick the methods (or even columns) you don't want run, which may be faster and easier to do if there are only some methods you don't want run.

This came up when trying to troubleshoot some issues pulling data that appears to be related to the get_peaking_factors and get_ts_parameters methods, and we would like to be able to just shut off these methods only.

gtrevisan, 2024-12-05:
hmm, I think at this stage you should rather get a list of all methods and drop what you do not want.

@amdecker, could you please share a snippet to accomplish this?
if this is not achievable easily, then we can try and implement it in a user-friendly way.

amdecker, 2024-12-10:
Here's a small example showing how to get physics methods for all but a few specified physics methods.
```python
from disruption_py.core.physics_method.runner import get_all_physics_methods
from disruption_py.settings.retrieval_settings import RetrievalSettings
from disruption_py.workflow import get_shots_data

from disruption_py.machine.cmod import METHOD_HOLDERS

all_physics_methods = get_all_physics_methods(METHOD_HOLDERS)
method_names = [method.__name__ for method in all_physics_methods]

# E.g. retrieve data for everything except get_peaking_factors and get_ts_parameters
method_names.remove("get_peaking_factors")
method_names.remove("get_ts_parameters")

retrieval_settings = RetrievalSettings(
    time_setting="efit",
    run_tags=[],
    run_methods=method_names,
)
shot_data = get_shots_data(
    tokamak="cmod",
    shotlist_setting=[1150805012, 1150805013, 1150805014],
    retrieval_settings=retrieval_settings,
    num_processes=1,
)
```

gtrevisan, 2024-12-10:
okay, so roughly the only required _addition_ is something like this -- I didn't test it, I just wanted to see the impact of this new block into an existing script:
```python
# exclude some methods
from disruption_py.core.physics_method.runner import get_all_physics_methods as gapm
from disruption_py.machine.cmod import METHOD_HOLDERS as MH
exclude = ["get_peaking_factors", "get_ts_parameters"]
methods = [m.__name__ for m in gapm(MH) if m not in exclude]
```
let us know if this works for your purpose!

AlexSaperstein, 2024-12-12:
It does!

I checked pulling data on dev for shot 1051123016 using `run_tags=['all']` and got a crash

[image]

But when I tried to pull data again using `run_tags=methods` with the code above it succeeded in pulling the rest of the data for the shot.

[image]

Sorry about the pics. I had trouble copy/pasting over fastx

gtrevisan, 2024-12-12:
great! I'll close this for now, but Amos and I added a small todo on his list, that is, implement a friendlier way to fetch the methods.

closed 2024-12-12 by gtrevisan