MIT-PSFC/disruption-py #379: Fix references for physics methods

gtrevisan, 2025-01-09:
in order to easily provide a physics-based reference for all our physics methods, we should provide a `References` section in the docstrings which link to the relevant physics-based PRs or external reading resources.

watch out! in order to be properly parsed by mkdocs, these references should be in the public physics method, not in its private counterpart.
this will make very clear who carried out which physics improvement and when, while the classic git blame approach would highlight countless code-based linting, logging, and refactoring changes.

we should first figure out the best format for mkdocs support: https://mkdocstrings.github.io/python/

eg: https://mit-psfc.github.io/disruption-py/usage/physics_methods/d3d_built_in_method_reference/#disruption_py.machine.d3d.physics.D3DPhysicsMethods.get_density_parameters

to test out mkdocs locally:
```
poetry install --with docs
poetry run mkdocs serve
```

gtrevisan, 2025-01-09:
so for that one, probably something like an unordered list:
```
"""
Get electron density from EFIT, then compute dn_dt and Greenwald_fraction.

References
----------
https://github.com/MIT-PSFC/disruption-py/blob/matlab/DIII-D/get_density_parameters.m
https://github.com/MIT-PSFC/disruption-py/issues/238
https://github.com/MIT-PSFC/disruption-py/pull/249
"""
```

at worst, we might have to start putting some metadata in the PRs themselves.

gtrevisan, 2025-01-09:
or maybe even more succinct eg:
```
"""
Get electron density from EFIT, then compute dn_dt and Greenwald_fraction.

References
----------
- original source: [get_density_parameters.m](https://github.com/MIT-PSFC/disruption-py/blob/matlab/DIII-D/get_density_parameters.m)
- issues: #[238](https://github.com/MIT-PSFC/disruption-py/issues/238)
- pull requests: #[249](https://github.com/MIT-PSFC/disruption-py/pull/249)
"""
```
which yields:
[image]

amdecker, 2025-01-21:
I like the references section! 

Another idea I had would be to use footnotes
[image]

```
Calculate the normalized beta
...
beta_n is defined as the total plasma beta (p/(B^2/2mu_0))
divided by Ip [MA]/(a*Bt).[^1]
...
[^1]: http://wiki.fusenet.eu/fusionwiki/index.php/Beta (accessed 1/21/25)
```

We could even have both footnotes and references where the references section would be for code/prs/issues and use footnotes for more direct citations of a paper or, in this case, FusionWiki.

amdecker, 2025-01-21:
Another way to improve the docs could be LaTeX integration: https://gitlab.com/mbarkhau/markdown-katex

gtrevisan, 2025-01-21:
yes! LaTeX support has been on the todo list for quite a while.

@amdecker if you feel you have time for this, please push a new branch with the necessary pyproject.toml and mkdocs.yaml updates, and maybe push an example for William to see.

@yumouwei let's then agree on something like `References` for the matlab original scripts and any notable issues, or PRs, while let's use footnotes for more manuscript-like hyperlinks, eg publications or fusion wikis.

yumouwei, 2025-01-27:
I like the footnote + reference link format. I also removed the "last major update by..." line since the info is included in the PRs.

[image]

yumouwei, 2025-01-27:
For methods that have both the underscored & the not underscored functions, I prefer to move the method descriptions which are often included in the docstring of the underscored function to the "main" function.

e.g. `get_ip_parameters`: instead of `Retrieve and interpolate Ip parameters.` now we have the follows. I believe this is more descriptive and useful for the users.

[image]

gtrevisan, 2025-01-27:
yes, of course! I believe only the public methods' docstrings automatically generate an API documentation page.