MIT-PSFC/disruption-py #382: Dumped CSV data don't match the output of get_shots_data

yumouwei, 2025-01-14:
# Problem

I was trying to compile a dataset using a list of shots, and I noticed that some signals in the csv file saved using the `output_setting=filename.csv` setting don't match those iFn the dataframe object output by the `get_shots_data()` function.

# Test case

(on disruption_py==v0.8.0)
[disruptionpy_data_mismatch.py.txt](https://github.com/user-attachments/files/18416111/disruptionpy_data_mismatch.py.txt)
[temp_shotlist.csv](https://github.com/user-attachments/files/18416112/temp_shotlist.csv)
(The script takes ~40s to run)

The outputs I got are:
[image]
[image]

gtrevisan, 2025-01-15:
oh, wow! thanks for reporting! definitely a nasty bug. :open_mouth: 

a couple of points.
- I imagine you might be running with the latest dev, but for future reference please report your actual version, not the last tagged version. eg: `git describe` or `git rev-parse HEAD`, and not v0.8.0;
- please include code as a code block, not an attachment! also, it'd be great to work towards a minimum reproducible example, removing all extra stuff.
- similarly, those columns might be text, not screenshots.
- clearly the data for `te_width` in 1st pic is the data for `li` in the 2nd pic. did you take a look at whether it's only a matter of garbled column headers?
- is it reproducible? if the bug appears for your N shots, does it appear for N-1 shots? what about N-2? ... what about 2, or 1?
- please try without multiprocessing, as well.
- did you try saving your dataframe by hand, and reading it by hand?
- did you try and take a look at the various csv writing calls to see if anything jumps at you? `git grep to_csv disruption_py/settings/output_setting.py`

my educated guesses are:
1. multiprocessing issue in saving batched csv data,
2. poor cycling capabilities of pandas to_csv/read_csv, which we do not properly take into account.

please, let me know! :pray:

yumouwei, 2025-01-17:
Here's an updated version of the example.

```
from disruption_py.settings import LogSettings, RetrievalSettings
from disruption_py.workflow import get_shots_data

import numpy as np
import pandas as pd

shotlist = [
            1160405017, 1160412004, 1160412021, 1160412028, 1160412031, 1160413005, 1160415006, 1160415011, 1160415016, 1160415023, 
            1160415025, 1160415030, 1160427016, 1160503007, 1160503009, 1160503011, 1160503015, 1160503020, 1160503023, 1160503026, 
            1160503027, 1160504006, 1160504007, 1160504009, 1160504010, 1160504012, 1160505018, 1160506005, 1160506016, 1160511003, 
            1160511006, 1160511007, 1160511008, 1160511009, 1160511011, 1160511012, 1160511013, 1160511014, 1160511015, 1160511016, 
            1160511017, 1160511018, 1160511019, 1160511020, 1160511022, 1160511023, 1160511025, 1160511027, 1160512001, 1160512002, 
            1160512003, 1160512004, 1160512005, 1160512006, 1160512007, 1160512008, 1160512016, 1160512020, 1160512023, 1160512027, 
            1160512030, 1160512032, 1160513004, 1160513005, 1160513006, 1160513007, 1160513008, 1160513009, 1160513010, 1160513012, 
            1160513013, 1160513014, 1160513015, 1160513016, 1160513017, 1160513019, 1160513020, 1160513022, 1160513027, 1160520006, 
            1160520018, 1160525006, 1160525019, 1160607003, 1160607018, 1160607026, 1160607030, 1160607031, 1160608008, 1160608015, 
            1160608017, 1160609002, 1160609005, 1160609013, 1160609026, 1160609027, 1160615010, 1160615019, 1160615023, 1160617008, 
            1160617015, 1160617016, 1160617018, 1160617021, 1160617022, 1160617027, 1160617033, 1160617034, 1160621003, 1160628002,     # Including this row (even only the first shot) causes the bug!
            1160628003, 1160628004, 1160628005, 1160628006, 1160628007, 1160628008, 1160706001, 1160707001, 1160707002, 1160707003, 
            1160707004, 1160707005, 1160707006, 1160707007, 1160707008, 1160707009, 1160707010, 1160707011, 1160707012, 1160707013, 
            1160707014, 1160707015, 1160708011, 1160708015, 1160708021, 1160712003, 1160712006, 1160712012, 1160712018, 1160713015, 
            1160714006, 1160714013, 1160714017, 1160714025, 1160714027, 1160714033, 1160718013, 1160720012, 1160720019, 1160720021, 
            1160721012, 1160721020, 1160722007, 1160722008, 1160725019, 1160725026, 1160728002, 1160728004, 1160728013, 1160728021, 
            1160728022, 1160728023, 1160729018, 1160803023, 1160803028, 1160803029, 1160803030, 1160803035, 1160804012, 1160804025, 
            1160804029, 1160805001, 1160805006, 1160805017, 1160809004, 1160809012, 1160809014, 1160810004, 1160810009, 1160811011, 
            1160811013, 1160811015, 1160816006, 1160816009, 1160816017, 1160816018, 1160816019, 1160816020, 1160816025, 1160816026, 
            1160817002, 1160817009, 1160818017, 1160819006, 1160819008, 1160823006, 1160823007, 1160823019, 1160824002, 1160824003, 
            1160824004, 1160824005, 1160824011, 1160824012, 1160824013, 1160824019, 1160824024, 1160825001, 1160826007, 1160826009, 
            1160826022, 1160826024, 1160826028, 1160831016, 1160831018, 1160901007, 1160901013, 1160901017, 1160902007, 1160902013, 
            1160902019, 1160907004, 1160907007, 1160907011, 1160907014, 1160908010, 1160908012, 1160908014, 1160908018, 1160908020, 
            1160908021, 1160909025, 1160909026, 1160915015, 1160915021, 1160916007, 1160916008, 1160920023, 1160921016, 1160922001, 
            1160922018, 1160922020, 1160922022, 1160922025, 1160922027, 1160922029, 1160922032, 1160923006, 1160923016, 1160923025, 
            1160926032, 1160928010, 1160928020, 1160929001, 1160929004, 1160929007, 1160929008, 1160929009, 1160929027, 1160929028, 
            1160929030, 1160930009, 1160930011, 1160930015, 1160930023, 1160930025, 1160930030, 1160930034, 1160930038, 1160930039,
            ]

# shotlist = shotlist[:100] # match
# shotlist = shotlist[:101] # mismatch

import random
random.seed(20) # Tested seeds: (N=101: Matched: 10, 15; Mismatch: 2, 11, 20, 100)
shotlist = random.sample(shotlist, 101) # mismatch occurs at N=101. changing num_processes does not affect the outcome

# Get flattop shots data
retrieval_settings = RetrievalSettings(
    efit_nickname_setting = "disruption",
    time_setting = "disruption_warning",
    domain_setting = 'flattop',
    run_tags=['all']
)

ram_df = get_shots_data(
    shotlist_setting = shotlist,
    tokamak = "cmod",
    retrieval_settings = retrieval_settings,
    output_setting = "dumped_data_v2.csv",
    num_processes = 1,
    log_settings=LogSettings(  # logging
        log_file_path=None,  # Defaults to "output.log" in the tmp folder for the session
        file_log_level="SUCCESS",
        log_file_write_mode="w",
        log_to_console=True,
        use_custom_logging=False,
    )
)

dumped_df = pd.read_csv('dumped_data_v2.csv')
[... 9 lines cut ...]
    b = dumped_df.iloc[mismatch_rows][['shot', 'time']+mismatch_columns]
    print(a.compare(b, keep_shape=True, keep_equal=True))

print(dumped_df['shot'].unique())
```

Let `N = len(shotlist)`:
- The bug may occur when `N > 100` (see my note on "tested seeds") 
- The columns that may have mismatched data are: `i_efc, n_e, dn_dt, greenwald_fraction, te_peaking, ne_peaking, pressure_peaking, te_width`. This contradicts the mismatched `li` column in the screenshots above; I did check `li` of that shot but the data do match each other.
[... cut ...]
- Importantly, changing `num_processes` does not change the outcome. The bug will still occur regardless using `num_processes=1` or `20`.

I have a feeling that this may have to do with MDSplus. Please try running this script and see if you can reproduce the same bug.

Other bullet points:

- disruption_py commit: `bf8820fc207457b2f2b6cfc51c8961a6d1f3e861`
- I tried saving and loading a dataframe manually using pandas. The loaded dataframe matches the original one.
```
ram_df.to_csv('temp.csv', index=False)
new_df = pd.read_csv('temp.csv')
np.isclose(ram_df, new_df).all()
> True
```

gtrevisan, 2025-01-30:
closed with:
- #389

closed 2025-01-30 by gtrevisan