MIT-PSFC/disruption-py #406: Smoothing v_loop and p_oh

AlexSaperstein, 2025-02-26:
`v_loop` can be a very noisy signal on C-Mod, which in turn makes the `p_oh` signal often very noisy as well. when the plasma has significant ohmically heating (`p_oh` >~ p_aux), this then leads to a very noisy `f_rad` signal which can interfere with disruption prediction analyses. An an example of this is shown below for shot 1120105021, where the original `v_loop`, `p_oh`, and `radiated_fraction` disruption-py signals are in blue. The bottom plot also shows the comparison between `p_oh`, `p_icrf`, and `p_lh` to show how `p_oh` is comparable to p_aux.

[image]

To mitigate this issue, I have been smoothing `v_loop` and then recalculating `p_oh` and `f_rad` from that smoothed signal. This is what is shown in the orange signals. Something similar could maybe be implemented in disurption-py.

gtrevisan, 2025-02-27:
thanks for the suggestion!

why don't you share a few snippets of your smoothing technique?
how sensitive is your desired result, in your experience, to the smoothing parameters?

this seems of broader interest than just p_oh, so maybe it's time to tackle how to setup generalized physics methods arguments.
not only in general, that is, passing arguments to physics methods, but also in the specific case of requiring signals to be smoothed over time.
let's table this for discussion at the next meeting!

AlexSaperstein, 2025-02-27:
Great!

The smoothing technique I use is just a simple box-car average with a 100ms window. Although it was just put together as a quick fix at the time, so something more sophisticated could also be used as well. The only requirement to consider is that the smoothing technique preserve causality. The box-car was just an easy way of implementing that, but there may be other better ones.

yumouwei, 2025-03-03:
I'm glad this one got brought up again. It is one of the things I'm planning to fix after #344 and #367.

So regarding vloop, there is another node (`efit_aeqdsk:vloopt`, currently called `v_loop_efit` in disruption-py) that is sampled at efit timebase and is smoother than the default `mflux:v0` used by the `get_ohmic_power` method. From the [efit documentation](https://nstx.pppl.gov/nstx/Software/Applications/EFIT_Signals.htm) this is supposed to be the measured loop voltage but I'd guess there's some additional smoothing done to it. 

[image]

Currently the `p_oh` computation uses `ip` from the magnetics, and `li`, `r0` from efit; by default we use the measured vloop (`\analysis:top.mflux:v0`), and if it's not available we then use `v_loop_efit` instead. The method first interpolate all of the signals to the requested time base and then compute `p_oh`. Note that there is this `dip_smoothed` signal which is calculated using `np.gradient` and `core.utils.math.smooth` (reimplementation of the matlab [smooth](https://www.mathworks.com/help/curvefit/smooth.html) function); this smooth function looks non-causal to me so we really need to fix it.

https://github.com/MIT-PSFC/disruption-py/blob/fbc1151c725ca4a820843aa66fd39393b94ea50b/disruption_py/machine/cmod/physics.py#L481-L487

https://github.com/MIT-PSFC/disruption-py/blob/fbc1151c725ca4a820843aa66fd39393b94ea50b/disruption_py/core/utils/math.py#L141-L145

To me it wouldn't be farfetched to switch to using `v_loop_efit` by default, as two of the 4 signals used in this calculation already come from efit. We can also do smoothing but we need to make sure that 1. the smoothing function is causal, and 2. we do not introduce any phase shift unintentionally. 

It'd also be curious to figure out how this `vloopt` signal was calculated.

yumouwei, 2025-03-10:
I was looking at applying causal boxcar filter to the raw `mflux:v0` vloop signal. Now I'm even more confused about how the efit `aeqdsk:vloopt` was obtained -- I have no idea how it obtained the data points around 0.1 s and 1.45 s.

[image]

Regarding the boxcar filtered signal: the 12-point filter does reduce the noise level by a lot, but it also introduces a delay of about 1 ms. What we should decide is how much we want to balance between noise level and delay.

[image]

For reference, here's the filtered signal using a 6-point window; the delay is now about 0.5 ms

[image]

[image]

yumouwei, 2025-03-10:
I was also looking at swapping the noncausal boxcar `smooth` function with this new causal boxcar filter to compute `dip_smoothed` (used in computing `p_oh`). The 6-point window filter also introduces a lag of about 0.5 ms.

[image]

yumouwei, 2025-03-14:
I had a discussion with Cristina the other day, and we thought the 0.5 ms delayed one would be better since we're planning to recompute EFIT with a uniform 1 ms interval. I'll put up a PR later today.

closed 2025-04-17 by yumouwei