MIT-PSFC/disruption-py #408: Improve shot domain identification

yumouwei, 2025-03-03:
Disruption-py identify the shot domain (ramp up, flattop, or ramp down) by checking 2 conditions:

- `abs(dipprog_dt) <= TRESHOLD_DIPPROG_DT`
- `abs(ipprog) > THRESHOLD_IP_PROG`

For CMOD, the values of these two thresholds are currently `TRESHOLD_DIPPROG_DT=1e3` and `THRESHOLD_IP_PROG=1e5`. Both Alex and I have found another set of threshold values which appear to be more robust than these default values; in my case I used `THRESHOLD_DIPPROG_DT = 5.0e4` and `THRESHOLD_IP_PROG = 1.0e5` in the ramping milestone project ([link](https://github.com/MIT-PSFC/onw_dev/blob/develop/onw_dev/generate_cmod_data/classify_disruption_domain.py) to script). We should take a look at this again and determine a best set of values.

In addition, I think it would be more useful to move domain identification from a setting to a physics method and returns the information as a categorical feature (e.g. 0=ramp up, 1=flattop, 2=ramp down). With this change the requested time base would be determined by `time_setting` alone, instead of by both `time_setting` and `domain_setting` as it's done currently. From a user's perspective, I'd say it's more convenient to store all data inside a single dataset and slice it to get the shot domain I'm interested in, instead of having 3 separate files and switch among them for each analysis.

gtrevisan, 2025-03-03:
I edited your comment for clarity, please double check the numbers -- you adopted the same IP threshold but you raised the DIP threshold by 50 times, right?

it would be helpful to see some traces here and there and compare the logic.

@crea-psfc and @granetz should comment on the physics when they can, while I will tackle the data aspect.
on the latter, I agree, preprocessing should be kept at a minimum and that might very well be an output step function or even an xarray attribute.
let's table this for discussion at our next meeting, too.

granetz, 2025-03-03:
@yumouwei @gtrevisan @crea-psfc 1 kA/s is the value I used for the C-Mod flattop constraint in the disruption warning database

crea-psfc, 2025-03-04:
@yumouwei I agree in turning IP phases identification into a categorical variable.
Did you check your thresholds work also on DIII-D time traces? Should set these in device configuration?

yumouwei, 2025-03-17:
I took a look at some of the shots from the dataset I compiled recently for this year's W&M summer school. From what I found (see slides below) I think `THRESHOLD_DIPPROG_DT=50e3` is better.

[CMOD flattop domain identification.pdf](https://github.com/user-attachments/files/19288878/CMOD.flattop.domain.identification.pdf)

gtrevisan, 2025-09-04:
closed with:
- #433

closed 2025-09-04 by gtrevisan