MIT-PSFC/disruption-py #415: Exception when shotlist_setting="plasmas" on Cmod

andrew-maris, 2025-03-13:
When trying to get all plasmas from Cmod, I get the below error that kills the data retrieval.

> 18:07:22.205 [CRITICAL] #930721017 | Failed to set up shot! %TREE-E-NODATA, No data available. Tree: analysis, Node: \analysis::efit:results:a_eqdsk:time

I have attached the full terminal output and error message below as a .txt. I am running on mfews04.

[terminal_cmod_plasmas_error.txt](https://github.com/user-attachments/files/19237557/terminal_cmod_plasmas_error.txt)

Below is the script:

```
#!/usr/bin/env python3

"""
Example usage of `get_shots_data` with all the default arguments explicitly assigned.
"""

from disruption_py.settings import LogSettings, RetrievalSettings
from disruption_py.workflow import get_shots_data

retrieval_settings = RetrievalSettings(
    cache_setting=None,
    efit_nickname_setting="default",#"default",#"disruption",
    # method/column selection
    # default None: all methods/columns
    # run_methods=None, #List of physics methods to run (default is None). If None, and run_columns is None, all methods will be run.
    # run_columns=None, # See above
    only_requested_columns=False,
    custom_physics_methods=[],
    # timebase settings
    time_setting="disruption_warning",#"efit",
    domain_setting="full",
    use_cache_setting_timebase=False,
    # not yet implemented
    interpolation_method="linear",
)

shot_data = get_shots_data(
    # required argument
    shotlist_setting="plasmas",#[1150805012, 1150805013, 1150805014],
    # default None: detect from environment
    tokamak="cmod",#None,#
    # default None: standard SQL/MDSplus connection
    database_initializer=None,
    mds_connection_initializer=None,
    retrieval_settings=retrieval_settings,
    output_setting="dataframe",
    num_processes=1,
    log_settings=LogSettings(
        # default None: "output.log" in temporary session folder
        log_file_path=None,
        file_log_level="DEBUG",
        log_file_write_mode="w",
        log_to_console=True,
        # default None: VERBOSE, or higher based on number of shots
        console_log_level=None,
        use_custom_logging=False,
    ),
)

# Save the shot data to a CSV file
shot_data.to_csv("cmod_shot_data.csv", index=False)
```

It looks like the issue is it's trying to get efit18 data for a shot that isn't in the tree. Is there a way to skip over shots like these?

yumouwei, 2025-03-17:
I checked jTraverser & dwscope, and it looks like that particular shot doesn't have EFIT data (in the analysis tree) at all. 

[image]

To my understanding, EFIT18 was only done for shots in Bob's shot list (so shots in the 105 and 112-116 ranges)

yumouwei, 2025-03-17:
That being said, I also tried to run `default.py` using:

- efit_nickname_setting='analysis'
- time_setting='efit'
- shotlist_setting=[941130007]  # We tested this shot in #332 and it does have `ANALYSIS.EFIT` data

but I'm getting an `MDSplus.mdsExceptions.TdiINVCLADSC: %TDI-E-INVCLADSC, Storage class not valid, must be scalar or array` error. I wonder what was changed since #332.

gtrevisan, 2025-03-18:
this issue boils down to a shot not having any EFIT information within MDSplus, and thus failing at an initial stage of the shot data retrieval process with a critical error.

for the future, please isolate the problem better, eg:
```
poetry run disruption-workflow -s 930721017
```
if that did not provide enough logging, you could also add `-l debug` or `-l trace`.
the above command simply runs the `simple.py` example for the shot at hand.

obviously there is nothing we could do for that specific shot.
- [ ] on one hand, we should investigate how come we have an otherwise-reasonable shot that does not have any EFIT tree. is this coming out of the "plasmas" shortcut query? should we update that? @yumouwei 
- [ ] on the other hand, the present issue then becomes: should we tolerate one shot fatally failing to be setup? the answer should be yes! I agree, I'll push a PR.

gtrevisan, 2025-03-20:
closed with:
- #418

closed 2025-03-20 by gtrevisan