MIT-PSFC/disruption-py #443: Fix get_end_of_shot method in time_setting

gtrevisan, 2025-04-16:
for some shots, no `valid_indices` are found in the `_get_end_of_shot` class method:

https://github.com/MIT-PSFC/disruption-py/blob/e4142156fa2f2237b251fc2d41f0589d23060d72/disruption_py/settings/time_setting.py#L462-L463

and thus we fail with:

```
ValueError: zero-size array to reduction operation maximum which has no identity
```

a couple of examples:
```
# on DIII-D
poetry run disruption-py 173927
# on EAST
poetry run disruption-py 59684
```

since this happens when setting up the timebase, such an exception is handled as critical and any further computation is impossible.

is there any fix? do these shots really deserve to be discarded?
otherwise we might just add a check on `valid_indices` and raise a user-friendlier exception.

yumouwei, 2025-04-18:
For the `ValueError`, the only fix needed seems to be adding a pair of parentheses in L446. I'm more surprised how this hasn't cause any trouble before.

https://github.com/MIT-PSFC/disruption-py/blob/17589bfc3edfa0480d045fa141666394ae0d0057/disruption_py/settings/time_setting.py#L446-L448

D3D shot 173927 looks really weird though. My guess is that the ip timebase was shifted by ~7s, and the EFIT software automatically stopped at 10s which is why there's no EFIT data in the second half of the shot .

[image]

yumouwei, 2025-04-18:
The other EAST shot looks a lot more usual. It seems the only problem was caused by that `np.where(signal_time<=0)` line, as `signal_time` started at 0.03 for this shot.

[image]

gtrevisan, 2025-04-22:
there are at least 3 cases in which, at a glance, I'd add a comma to unwind the tuple returned by `np.where`.
would you be able to check this out and fix all logic within that function? maybe in #439?

gtrevisan, 2025-05-07:
closed with:
- #439

closed 2025-05-07 by gtrevisan